fieldOfView/Cura-ZOffsetPlugin #2: There should not be a minimum value

aguadopd, 2018-05-20:
Currently, we have 
```python
"minimum_value": "-layer_height_0",
```

For accurate bed leveling I wan to use a credit card, which has a 0,8mm height. I need to apply a negative offset bigger than my layer height in order to get to it. Maybe using a _minimum value warning` instead of a hard limit?

PS: Thanks for the plugin, it's helping me a lot with calibration.

fieldOfView, 2018-05-20:
Specifying a value lower than negative the Initial Layer Height would result in a negative Z value in the gcode. The firmware would - at best - ignore the negative value, or give you an error.

aguadopd, 2018-05-20:
Is this because of the way the plugin works, or something firmware related? Could I solve this with a custom start Gcode instead of using the plugin?

fieldOfView, 2018-05-20:
The firmware assumes that Z=0 means that the nozzle touches the buildplate. A negative Z would mean that the nozzle is below the top surface of the buildplate (ie: the nozzle pushes through the buildplate). Needless to say, the nozzle actually pushing through the buildplate would be "less than desirable".

aguadopd, 2018-05-20:
Well, that makes a lot of sense. I've just remembered a nice calibration procedure I saw in one custom printer with Marlin: you could set a calibration height and then it started going through the bed. Mesh bed levelling, I recall.
I shall do something like that. Maybe running a custom GCode after manually setting the desired height (the feeler height). Manual operation through the LCD is cumbersome and PC connection is not available.

Thanks a lot for your comments. I close this issue.

closed 2018-05-20 by aguadopd