fieldOfView/Cura-ZOffsetPlugin #7: Non-zero value results in printhead starting ~75mm up the Z axis

SnowDrifterr, 2019-11-09:
I installed the z-offset plugin from the market place, set it to 1.7mm(imported from S3D), and its behavior is to attempt to print ~75mm off the print bed. Seem to happen on any non-zero setting. When loading the preview in flashprint, the print does show on the bed.

Printer is a flashforge adventurer 3

Setting to zero, or uninstalling the plugin results in normal printing behavior. But I still need to be able to adjust it for my use case

SnowDrifterr, 2019-11-10:
First couple lines of g-code:

;FLAVOR:Marlin
;TIME:5183
;Filament used: 1.13638m
;Layer height: 0.1
;MINX:-26.655
;MINY:-14.893
;MINZ:0.2
;MAXX:26.658
;MAXY:14.881
;MAXZ:4.4
;ZOFFSETPROCESSED
;Generated with Cura_SteamEngine 4.3.0
M140 S40
M105
M190 S40
M104 S240
M105
M109 S240
M82 ;absolute extrusion mode
M140 S[bed0_temperature] ; set bed temperature
M104 S[extruder0_temperature] T0 ; set extruder temperature
M104 S0 T1 ; set extruder temperature
G28 ; home all axes
M132 X Y Z A B ; recall home offsets
G1 Z50 F3300 ; lower bed
M7 T0 ; stabilize bed temperature
M6 T0 ; stabilize extruder temperature
;M651 ; set case fan on
M907 X100 Y00 Z40 A100 B20 ; set motor current
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:43
;LAYER:0
M107
G0 F3600 X-.753 Y14.749 Z1.9 ;adjusted by z offset
G92 Z0.2 ;consider this the original z before offset
;TYPE:SKIRT
G1 F1500 E0
G1 F600 X-4.279 Y14.868 E0.10267
G1 X-4.511 Y14.869 E0.10943

fieldOfView, 2019-11-11:
As pointed out by @Ghostkeeper on reddit:

> The Z offset code seems to be working fine by that g-code. It moves first to Z1.9 and then says that the current coordinate is Z0.2. That effectively results in an offset of 1.7mm.
> 
> There's one code in your starting g-code that looks weird. It's this one:
> 
> M132 X Y Z A B ; recall home offsets
> 
> I'm not familiar with this Makerbot-specific g-code. But according to the documentation on Reprap.org it could be the thing that's messing with your Z offset. Depends what's stored in that EEPROM. But an EEPROM can get corrupt or just fail to read.

Basically, this is your firmware not interpreting gcode according to its specification. Interestingly it does so in a way that is comparable to the newer Ultimaker printers. I am special-casing these printers to use a different (far less efficient) way to offset the prints. I may expose that alternative way in a future version of the plugin.

fieldOfView, 2019-12-16:
I have submitted a version of the plugin to the Marketplace, that has a new setting named "Extensive Z Offset Processing" (disabled by default). If you enable that setting, the plugin applies its offset to the whole gcode file instead of adjusting its coordinate system.

It may take a couple of days before the new version is approved for the marketplace. If you cannot wait, you can download a Cura 4.4 version of the plugin here:
https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/ZOffsetPlugin/0rMCzAbGG-0MKX0jHgnI4tx1XkZJyHI9TzY-M8Ax70YK/ZOffsetPlugin-v7.0.0-2019-12-16T16%3A12%3A46Z.curapackage?Expires=1891873284&GoogleAccessId=stardust-storage-production%40stardust-193112.iam.gserviceaccount.com&Signature=kENBRbLDSHxjxfC5I9UxrKTPI98rmQyruA9K3NUCQJ2k9FOU%2BZSLq%2FOOEkdzw3OpEzAeGpzJfizHHImpIhnmrDRQCXNQvPSBIp9HTeZ4yBGhefz8FelaZxJVHeD7Ips3rKNMMH30pVQZcrmulkkuArwCXm3nLXVLHemsYzd%2F9Kf9Z5zeIc%2FNh%2B3GDPX%2FvK2%2B8tN0eDtXHpzVz6mD6Dl5YhJm1CKzeJcn9ieDsDdk2QbRoWwBHrSvzpYBFLRmXGM3Fyaef60FKs47yslyPdMlaC8xRLO4C%2BGhXBqMXydHHUAK%2BJoeAmoShxUb8sUTPSKvyuVR6WoLQucgQFymTTtJDw%3D%3D

closed 2019-12-16 by fieldOfView