fieldOfView/Cura-ZOffsetPlugin #9: Doesn't seem to be enabled in Cura 4.4

agibson2, 2019-12-10:
I assume it isn't compatible with 4.4.  If I go to the installed plugins and check it, it just goes back to being unchecked when I restart Cura.  This is a very useful feature in Cura.  It saves me from needing to use the very long and inaccurate bed leveling procedure built into my printer.

fieldOfView, 2019-12-10:
The version of the plugin that is available in the Marketplace is compatible with Cura 4.4. What version do you have installed? You can check this on the "Installed" tab of the Marketplace.

agibson2, 2019-12-11:
3.5.4 11/8/2019
[image]

agibson2, 2019-12-11:
I also see this in the Cura logs...
2019-12-10 21:10:20,886 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [374]: Plugin [ScalableExtraPrime] with supported sdk versions [['4.0.0']] is incompatible with the current sdk version [7.0.0].
2019-12-10 21:10:20,946 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [374]: Plugin [ZOffsetPlugin] with supported sdk versions [['6.0.0']] is incompatible with the current sdk version [7.0.0].

agibson2, 2019-12-11:
Don't spend any time looking at this.  I just deleted Cura 4.4 and reinstalled it.  It didn't import my 4.3 config which is still there (as well as 4.3 Cura) but that is ok.  I am taking advantage of the time to just start from scratch on the configs.  The Z Offset is working with the new install from scratch.

closed 2019-12-16 by fieldOfView