fieldOfView/Cura-ZOffsetPlugin #12: Cura 4.5.0 will not save gcode when z offset used

StanislavJochman, 2020-04-06:
**Application version**
4.5.0
**Platform**
Manjaro 19.0.2 Kyria
Kernel: x86_64 Linux 4.19.113-1-MANJARO
**Printer**
Does not matter

**Reproduction steps**
1. Installed z offset
2. Set z offset to 0.1mm
3. Sliced Gcode - everything looks fine but can not save gcode
4. Error in terminal see screenshot

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]
**Logs**
2020-04-06 13:45:35,281 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [144]:     adjusted_z = round(float(result.group(2)) + z_offset_value, 5)
2020-04-06 13:45:35,281 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [144]: ValueError: could not convert string to float: ''
2020-04-06 13:45:39,218 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles

**Additional information**
If z offset is set to 0 it works fine but when it is set to 0.1 and I use 0.8 mm nozzle it throws this error

fieldOfView, 2020-04-06:
Can you post a gcode file that is saved when the z offset is set to 0? I'm having a bit of a hard time figuring out how this could happen.

StanislavJochman, 2020-04-06:
It posted normal G code which it posts even without the plugin.  Some even number for example layer height set as z offset works but 0.1mm does not when layer height is set to 0.55mm with 0.8mm nozzle.

StanislavJochman, 2020-04-06:
It looks like you are trying to convert empty string to float but I do not know why I have not investigated yet.

fieldOfView, 2020-04-06:
I am asking for a gcode file without the plugin enabled, so I can see what causes the empty string being used instead of a coordinate. Can you post a (small) gcode file so I can see why finding the z coordinate would fail?

fieldOfView, 2020-04-06:
I think I know what's happening. The plugin does not like negative Z coordinates, and they could happen if you use relative coordinates in a part of the gcode file. Could you post your start and end gcode snippets?

fieldOfView, 2020-04-06:
Could you try this [development snapshot](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/ZOffsetPlugin/wH6PWz5jbylXGkAZ2enW0TK_3DtHLFPGGU6StTX7Jcx0/ZOffsetPlugin-v7.0.0-2020-04-06T16%3A24%3A58Z.curapackage?Expires=1901550301&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=Tefly3xxheL7PJvtqkg0RgW%2FYyHB28a3hUj6DxDCTLSNYoAeOKI0Y7zpMfuAUZtGyXaq8Gx1QjklwRuUMjLOI9jzP194dvsYuVzoRt%2F0eEPqj7Jtyr4o5AUev3VwBrVgtRpXGdjl4Fdgg1KRYVtyMELBXXUmxFk6Q0iFoYgpA4C18DJQhorarmAH%2BugqD4FOoi4WDvpN9kKAkKMVNfC4f1YpAEjpSJ364GiKRH8l2iUThvrD2zlaHgtYyUvAOT92%2FtugGMOHVO6YMMVWSucTb9rf%2BTMocoZ5nONMDqe7C2HSLVWeATfI3upvhxqkW4lpN8vYxHgAuuDRFHn4fMfIiw%3D%3D)? Download the file and drag it into Cura as if you were opening a model. Then restart Cura as instructed.

StanislavJochman, 2020-04-07:
> I think I know what's happening. The plugin does not like negative Z coordinates, and they could happen if you use relative coordinates in a part of the gcode file. Could you post your start and end gcode snippets?

I am posting my start and end Gcode:
**Start Gcode**
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
M140 S{material_bed_temperature} ; start preheating the bed
M104 S{material_print_temperature} ?T0 ; start preheating hotend
M190 S{material_bed_temperature} ;wait to heat bed
M109 S{material_print_temperature} ?T0 ;wait to heat bed hotend
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
;M420 S1;restore manual mesh leveling
G92 E0 ;zero the extruded length
G1 Z0.8 F1000.0 ; go up
G1 F200 E10;extrude 10mm of feed stock
G92 E0 ;zero the extruded length again
G1 Y-3.0 F1000.0 ; go outside print area
G1 X30.0 E9.0  F1000.0 ; intro line
G1 X60.0 E12.5  F1000.0 ; intro line
G1 F9000
M117 StanislavJochman

**End Gcode**
M104 S0 ;extruder heater off
M140 S0 ;heated bed heater off (if you have it)
G91 ;relative positioning
G1 E-1 F300  ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more
G90 ;absolute positioning
G1 X0 Y200 F3600 ;move extruder out of the way by moving the baseplate to the front for easier access to printed object
M84 ;steppers off

StanislavJochman, 2020-04-07:
> I am asking for a gcode file without the plugin enabled, so I can see what causes the empty string being used instead of a coordinate. Can you post a (small) gcode file so I can see why finding the z coordinate would fail?

I am also sending you short Gcode - XYZ calibration cube with plugin disabled.
[xyzcube.gcode.txt](https://github.com/fieldOfView/Cura-ZOffsetPlugin/files/4447174/xyzcube.gcode.txt)

fieldOfView, 2020-04-08:
> End Gcode
> (...)
> G1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more

The problem is the +0.5. My plugin is not expecting the +. I am not sure if it is strictly legal to have that + there, but I will make my plugin robust against it.

StanislavJochman, 2020-04-08:
Ok I will remove that. Can I close this issue? Will the + sign be fixed in the future?
Best regards Stanley

closed 2020-05-13 by fieldOfView