fieldOfView/Cura-ZOffsetPlugin #13: Z-offset amount does not affect Cura's default start code

whoisak, 2020-04-28:
i'm not sure if upgrading to the latest version of Cura changed anything, but around the same time, I started having problems with the default line Cura draws in its start code - it drags against the bed and is way too close. i can edit the gcode or have separate printers, but between printing with and without painter's tape, this is annoying. would be nice if there was an attribute i could use in gcode, or the plugin considered the start/end gcode cura uses.

fieldOfView, 2020-04-28:
You can use the replacement pattern `{adhesion_z_offset}` in your start gcode snippet.

whoisak, 2020-04-29:
Thank you for such a quick reply. I very much appreciate it. 

It would look something like this?

`G1 X0.1 Y200.0 Z{adhesion_z_offset}+.3 F1500.0 E15`

Can I add to it like that?

fieldOfView, 2020-04-29:
No, you cannot use math like that, Cura will only replace the string `{adhesion_z_offset}` by the number representing the value (eg 0.15), so you would end up with `G1 X0.1 Y200.0 Z0.15+.3 F1500.0 E15` and your printer does not understand that.

You can play with relative coordinates though, eg:
```
G1 X0.1 Y100.0 Z0.3 ; move to start of prime line
G91 ; relative positioning
G1 Z{adhesion_z_offset} ; move up by {adhesion_z_offset} mm
G90 ; back to absolute positioning, Z is now 0.3 + {adhesion_z_offset} mm
G1 X0.1 Y200.0 F1500.0 E15 ; extrude prime line
```

closed 2020-04-29 by fieldOfView

whoisak, 2020-04-30:
I came up with the following alternate Cura start gcode using relative coordinates for the entire purge line. I think this would be useful for new people visiting this repo, maybe something to add to the read.me?

It reproduces the exact line, but uses the z offset to calculate it...

`; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
G1 E15 F2700.0; extract filament to zero out
G92 E0 ; Reset Extruder
G28 ; Home all axes

G91 ; set to relative
G1 Z{adhesion_z_offset} ; use z offset from cura

G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed (abs: X0 Y0 Z2.0)
G1 X0.1 Y20 Z-1.8 F5000.0 ; Move to start position (abs: X0.1 Y20 Z0.2)
G1 Y180 F1500.0 E15 ; Draw the first line (abs: X0.1 Y200 Z0.2 E15)
G1 X0.3 F5000.0 ; Move to side a little (abs: X0.4 Y200 Z0.2)
G1 Y-180 F1500.0 E15 ; Draw the second line (abs: X0.4 Y20 Z0.2 E30)
G92 E0 ; 	Reset Extruder
G1 Z1.8 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed (abs: X0.4 Y20 Z2.0)
G1 X4.6 Z-1.2 F5000.0 ; Move over to prevent blob squish (abs: X5.0 Y20 Z0.8)
G90 ; set to absolute`