fieldOfView/Cura-ZOffsetPlugin #14: Crash cura on startup

shep89, 2020-05-02:
Just after installing plugin from marketplace getting error
Tried on Cura 4.5 and 4.6.1
```
$ ./Ultimaker_Cura-4.6.1.AppImage 

(AppImageLauncher:26012): GLib-GIO-CRITICAL **: 22:55:59.196: g_converter_convert: assertion 'outbuf_size > 0' failed

(AppImageLauncher:26012): Gtk-CRITICAL **: 22:55:59.196: Unable to load resource for composite template for type 'GtkDialog': Не удалось распаковать ресурс из «/org/gtk/libgtk/ui/gtkdialog.ui»

(AppImageLauncher:26012): Gtk-CRITICAL **: 22:55:59.196: gtk_widget_class_bind_template_child_full: assertion 'widget_class->priv->template != NULL' failed

(AppImageLauncher:26012): Gtk-CRITICAL **: 22:55:59.196: gtk_widget_class_bind_template_child_full: assertion 'widget_class->priv->template != NULL' failed

(AppImageLauncher:26012): Gtk-CRITICAL **: 22:55:59.196: gtk_widget_class_bind_template_child_full: assertion 'widget_class->priv->template != NULL' failed

(AppImageLauncher:26012): Gtk-CRITICAL **: 22:55:59.196: gtk_widget_class_bind_template_child_full: assertion 'widget_class->priv->template != NULL' failed

(AppImageLauncher:26012): Gtk-CRITICAL **: 22:55:59.196: gtk_widget_class_bind_template_callback_full: assertion 'widget_class->priv->template != NULL' failed
QApplication: invalid style override passed, ignoring it.
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number
Fatal Python error: Cannot recover from stack overflow.

Thread 0x00007f86535e3700 (most recent call first):
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 549 in wait
  File "/home/shep/.local/share/cura/4.6/plugins/MKSWifiPlugin/MKSWifiPlugin/MKSOutputDevicePlugin.py", line 211 in _handleOnServiceChangedRequests
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f8653fff700 (most recent call first):
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 161 in _storeStderrToLogThread
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f8670f2b700 (most recent call first):
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 149 in _storeOutputToLogThread
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f8671ffb700 (most recent call first):
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 123 in run
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f86727fc700 (most recent call first):
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 123 in run
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f8672ffd700 (most recent call first):
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 293 in wait
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 424 in acquire
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 123 in run
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007f86737fe700 (most recent call first):
  File "/build/env/4.6/inst/lib/python3.5/threading.py", line 293 in wait
[... 86 lines cut ...]
  File "/home/shep/.local/share/cura/4.6/plugins/ZOffsetPlugin/ZOffsetPlugin/ZOffsetPlugin.py", line 53 in _onContainerLoadComplete
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 332 in __performEmit
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 219 in emit
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Settings/ContainerRegistry.py", line 210 in findContainers
  File "/home/shep/.local/share/cura/4.6/plugins/ZOffsetPlugin/ZOffsetPlugin/ZOffsetPlugin.py", line 53 in _onContainerLoadComplete
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 332 in __performEmit
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 219 in emit
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Settings/ContainerRegistry.py", line 210 in findContainers
  File "/home/shep/.local/share/cura/4.6/plugins/ZOffsetPlugin/ZOffsetPlugin/ZOffsetPlugin.py", line 53 in _onContainerLoadComplete
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 332 in __performEmit
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 219 in emit
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Settings/ContainerRegistry.py", line 210 in findContainers
  File "/home/shep/.local/share/cura/4.6/plugins/ZOffsetPlugin/ZOffsetPlugin/ZOffsetPlugin.py", line 53 in _onContainerLoadComplete
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 332 in __performEmit
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 219 in emit
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Settings/ContainerRegistry.py", line 210 in findContainers
  File "/home/shep/.local/share/cura/4.6/plugins/ZOffsetPlugin/ZOffsetPlugin/ZOffsetPlugin.py", line 53 in _onContainerLoadComplete
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 332 in __performEmit
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 219 in emit
  File "/build/4.6/build/inst/lib/python3.5/site-packages/UM/Settings/ContainerRegistry.py", line 210 in findContainers
  File "/home/shep/.local/share/cura/4.6/plugins/ZOffsetPlugin/ZOffsetPlugin/ZOffsetPlugin.py", line 53 in _onContainerLoadComplete
  ...
Aborted

```

fieldOfView, 2020-05-03:
Hey, I'm sorry you were hit by that bug. I have already fixed the bug, but have not yet submitted the fixed version to the marketplace because thankfully the bug does not manifest often.

The crash actually only happens on configurations that contain specifically broken files that cannot be successfully read back in. My plugin then triggers an infinite loop bug in Cura. The same bug also affects two other plugins of mine (Linear Advance Setting and Printer Settings).

Let me know if you need help getting your Cura back in working order. Removing the `/home/shep/.local/share/cura/4.6/plugins/ZOffsetPlugin` folder should work.

shep89, 2020-05-03:
Thank you for the answer. Please let me know if I can help somehow.

On the other hand could your please share the working files of the plugin? I am going to replace it manually.

fieldOfView, 2020-05-03:
You can download this [prerelease version](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/ZOffsetPlugin/wH6PWz5jbylXGkAZ2enW0TK_3DtHLFPGGU6StTX7Jcx0/ZOffsetPlugin-v7.0.0-2020-05-03T08%3A17%3A41Z.curapackage?Expires=1903854365&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=hlOlhv8WzYQ%2BQkZsKjuCVCxJ%2FkEay9u27%2BXFrBqeB0RFx65NassiAIkNjw17Ka0RqpTutosCOnWA7OU1lcS%2Fc4CZuof7Lt9JU6xy8xIwNKJ0ngstlZoekVUg27CedpI%2F8E9O45DDJL2uSIlAtMfGkUpFWLZJ%2B8kJbr01A25%2FewgbfYEOnDlOUK36OwBVnBBWcYfF%2FylA2rpWKEKMjZQfPtGCtO0W%2BFdeWkEVJCfOJD0W7uNDyYgwfC%2FtASsfI%2BWYRWxJHJ1kIvFoypBwp9InNI%2BID2%2FEMU4WYc8fAVnW2FGfg9GjuMcyqYdGZjXdxws40TOOKdzSv1%2FfdMq%2Brk4d6w%3D%3D) and drop it onto the buildplate in Cura as if you were opening a model. If that does not work, please try the files from the master branch: https://github.com/fieldOfView/Cura-ZOffsetPlugin/tree/master

shep89, 2020-05-03:
prerelease version works!

closed 2020-05-13 by fieldOfView