fieldOfView/Cura-ZOffsetPlugin #15: 3.5.6 Problem

Mitsch485, 2020-05-12:


fieldOfView, 2020-05-13:
I'm looking in to a fix.

fieldOfView, 2020-05-13:
Could you try [this version](https://storage.googleapis.com/ultimaker-cura-packages/v1/packages/ZOffsetPlugin/FXqHOcP2ClnA28sY9Idc7cw5AgEKDxWxWKzX0xqGQJgb/ZOffsetPlugin-v7.0.0-2020-05-13T06%3A33%3A55Z.curapackage?Expires=1904711639&GoogleAccessId=cura-packages%40stardust-193112.iam.gserviceaccount.com&Signature=gZtdeQ1d0h7Y9kWF3N4wMucu%2B7f6b%2BSqNmpLgzoj3PUWYY1NnJZI%2BTBs9YzVCBQK25x%2BOsrALPQPPMpsO4183Knd%2BYZvLuX8oHXmO8pbh94BtuLq9PDVwuVSRHmDYhxHr3YviG5%2B4775N160hgt7YJOHKdQ6hoEE9LKbRxGW7hp1R%2BH2zN8Weg%2FapmLQR91PAW585ugzWWY2hGlzlGmo1fNDqaV8VdzbzGqUW8Y7a2iWVwkEPvCvC2jv0J2%2BshAswoVokwzv9DbURwD4eG%2BkWW8WBp84WCW7SaYC23EIX9sgIyrhh7%2BZ58BNr44ddWFrAx%2Bv8R7vsDTonz172NX5Dw%3D%3D)? Download the file and drag it onto the Cura buildplate as if you were opening a 3d model.

Mitsch485, 2020-05-13:
Hi seems that now it is working again.

 

G1 F2100 E-1.5
;LAYER_COUNT:450
;LAYER:0
M107
G1 F300 Z0.5 ;adjusted by z offset
G92 Z0.7 ;consider this the original z before offset
G0 F6000 X145.239 Y120.432 Z0.7
;TYPE:SKIRT
G1 F300 Z0.2

 

 

 

 

But i will try it in a print later on.

Thank you very much

 

Gesendet: Mittwoch, 13. Mai 2020 um 08:35 Uhr

commit ebd12a9f (2020-05-13): Fix regular expression

Fixes #15
closed 2020-05-13 by fieldOfView