fieldOfView/Cura-ZOffsetPlugin #16: After installation on Cura 4.7 cura crashes continuously

levidehaan, 2020-09-02:
Cura crashes after loading Any stl file.

Not exactly sure how this happens, i've looked through the code and can't see where it happens yet, but it happened on both windows 10 and mac. 
I am also unable to reinstall as it makes a change Somewhere that stays even after uninstallation of Cura 4.7 and reinstallation on both mac and windows.
Cura was working fine on both Windows 10 and Mac until i installed this plugin.
Any chance someone else is having this issue?

Traceback (Mac):
Traceback (most recent call last):
  File "/Users/ultimaker/build/4.7/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 1060, in event
  File "/Users/ultimaker/build/4.7/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 464, in event
  File "/Users/ultimaker/build/4.7/build/inst/lib/python3.5/site-packages/UM/Event.py", line 218, in call
  File "/Users/ultimaker/build/4.7/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 332, in __performEmit
  File "/Users/ultimaker/build/4.7/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 1820, in _readMeshFinished
IndexError: list index out of range

Logs(Mac):
Thread 0x00007000127f8000 (most recent call first):
  File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000122f5000 (most recent call first):
  File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700011df2000 (most recent call first):
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/th3m4d0n3/Library/Application Support/cura/4.7/plugins/OctoPrintPlugin/OctoPrintPlugin/python-zeroconf/zeroconf/__init__.py", line 2184 in wait
  File "/Users/th3m4d0n3/Library/Application Support/cura/4.7/plugins/OctoPrintPlugin/OctoPrintPlugin/python-zeroconf/zeroconf/__init__.py", line 1496 in run
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000118ef000 (most recent call first):
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/th3m4d0n3/Library/Application Support/cura/4.7/plugins/OctoPrintPlugin/OctoPrintPlugin/python-zeroconf/zeroconf/__init__.py", line 2184 in wait
  File "/Users/th3m4d0n3/Library/Application Support/cura/4.7/plugins/OctoPrintPlugin/OctoPrintPlugin/python-zeroconf/zeroconf/__init__.py", line 1315 in run
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000113ec000 (most recent call first):
  File "/Users/th3m4d0n3/Library/Application Support/cura/4.7/plugins/OctoPrintPlugin/OctoPrintPlugin/python-zeroconf/zeroconf/__init__.py", line 1221 in run
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700010ee9000 (most recent call first):
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/site-packages/zeroconf/__init__.py", line 2117 in wait
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/site-packages/zeroconf/__init__.py", line 1434 in run
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000109e6000 (most recent call first):
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 549 in wait
  File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007000104e3000 (most recent call first):
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/site-packages/zeroconf/__init__.py", line 2117 in wait
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/site-packages/zeroconf/__init__.py", line 1262 in run
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

[... 78 lines cut ...]
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Current thread 0x0000000116770dc0 (most recent call first):
  File "/Users/ultimaker/build/4.7/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 393 in _logInfoWidget
  File "/Users/ultimaker/build/4.7/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 182 in _createDialog
  File "/Users/ultimaker/build/4.7/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 86 in __init__
  File "/Users/ultimaker/build/4.7/build/inst/bin/cura_app.py", line 166 in exceptHook
  File "/Users/ultimaker/build/4.7/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 378 in exec_
  File "/Users/ultimaker/build/4.7/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 844 in run
  File "/Users/ultimaker/build/4.7/build/inst/bin/cura_app.py", line 234 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/Users/ultimaker/build/env/4.7/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

fieldOfView, 2020-09-02:
This seems to be the same issue as https://github.com/Ultimaker/Cura/issues/8235 and https://github.com/Ultimaker/Cura/issues/8230. This should be fixed with Cura 4.7.1 which will be released somewhere this week.

fieldOfView, 2020-09-21:
Cura 4.7.1 is now available, fixing the crash above.

closed 2020-09-21 by fieldOfView