fieldOfView/Cura-ZOffsetPlugin #17: Offset Issue with Cura 4.7.1

hparHelios, 2020-10-07:
Using a Z offset sets the offset incorrectly. I used a 0.1mm offset and it printed at a couple of centimeters which caused the print to fail.

muenznej, 2020-11-23:
i am using it for 4.7 and its option is grey ( i think inactive ), dont know why

fieldOfView, 2020-11-23:
@hparHelios, I am sorry I totally forgot about this issue. What printer do you use? Have you tried using the "Extensive Z Offset Processing" option?

@muenznej, this sounds like a completely different problem than what @hparHelios is describing. Please could you open a new issue instead of attaching your problem to this one?

hparHelios, 2020-11-30:
> @hparHelios, I am sorry I totally forgot about this issue. What printer do you use? Have you tried using the "Extensive Z Offset Processing" option?
>
> @muenznej, this sounds like a completely [...]

fieldOfView,

Apologies for the delay as I've been out of the office this past week. I am using the Ultimaker S5 printer.

fieldOfView, 2021-02-05:
I can't reproduce this. If it is still happening, can you please post a link to a gcode or ufp file so I can inspect the gcode that is created? What firmware version do you have on your S5?

fieldOfView, 2021-02-23:
I can't fix this without further info. If you can post a link to a gcode or ufp file so I can inspect the gcode that is created I will reopen the issue.

closed 2021-02-23 by fieldOfView