fieldOfView/Cura-ZOffsetPlugin #18: zoffset setting grey / inactive

muenznej, 2020-11-23:
Hello,
since i switched from Cura 4.6 on windows to Cura 4.7.0 on linux
the offset plugin does not work for me anymore. its option is grey

_Originally posted by @muenznej in https://github.com/fieldOfView/Cura-ZOffsetPlugin/issues/17#issuecomment-732228054_

fieldOfView, 2020-11-23:
Thanks for adding the screenshot. The setting is not grey because it is disabled (it isn't), but it is grey because it is not set to be visible by default. Right-click the setting and select "Keep setting visible", and it will no longer be rendered grey if you search for it.

muenznej, 2020-11-23:
ow, sorry -.-
I had leveling problems and i blamed that option, thanks for poiting it out :+1: 
nice support :)

closed 2020-11-23 by muenznej

fieldOfView, 2020-11-23:
It is always nice to get the blame for something random ;-)

Anyway, thanks for reviving that other issue.

muenznej, 2020-11-23:
> t is always nice to get the blame for something random ;-)
programmers life :man_shrugging: