fieldOfView/Cura-ZOffsetPlugin #19: Z-Offset incrementing in some situations?

molodeztom, 2021-03-20:
Hi, thanks for a really helpful plug-in. 
Printer Ancubic i3 mega s, BL-Touch , Cura 4.8.0, Octoprint
I can adjust the distance nozzle to bed by applying e.g Z-offset -0.1. It works fine as expected. 
When the print finishes and I do the same print again starting from Cura it seems the z-offset is subtracted again. Like Z-offset -0.2.
So the nozzle is closer to the bed. 
It does not happen when the printer is switched off. It does not happen when I do auto bed leveling. 
Is this possible or just my feeling? Can I get the actual z-offset by a marlin command ?

fieldOfView, 2021-03-21:
Does your start gcode include a G28 home command for the Z axis? If it doesn't, the effect would indeed be incremental.

You can try setting the "Extensive Z Offset Processing" option. This will change all the Z coordinates in the gcode instead of manipulating the coordinate system.

akfpv-Nightprowler, 2022-04-08:
Hello, 
I‘m using the Plugin on to Printers with MBL. Cura 4.11 and 4.12 and habe the same problem. The second print fails because of bad first layer. G28 is part of the Startcode. I will test the extensive mode.

fieldOfView, 2022-04-25:
Did you get round to test the "Extensive Z Offset Processing" option?

fieldOfView, 2022-05-03:
I'm going to assume that using the "Extensive Z Offset Processing" option fixed it.

closed 2022-05-03 by fieldOfView