fieldOfView/Cura-ZOffsetPlugin #20: Cannot see the plugin it the settings

ThomasZett, 2021-04-15:
Cannot see the plugin it the settings on Cura 4.8
It is in 
C:\Users\Thomas\AppData\Roaming\cura\4.8\plugins\Cura-ZOffsetPlugin-3.5\Cura-ZOffsetPlugin-3.5

AND in C:\Program Files\Ultimaker Cura 4.8.0\plugins\Cura-ZOffsetPlugin-3.5

br, Thomas

fieldOfView, 2021-04-16:
Could you please try installing the plugin via the Marketplace instead of manually? Remove both the versions you have installed now.

Is the plugin listed in the Installed tab of the Marketplace?

To further diagnose the problem, I need to see your Cura.log. It can be found in %APPDATA%\cura\4.8\cura.log. Please post it somewhere and post a link here (don't paste the whole thing here, as it will be long).

ThomasZett, 2021-04-16:
Installing via Marketplace didi the trick!

Thank you very much for the fast help (and the plugin).

br, Thomas

closed 2021-04-16 by ThomasZett