fieldOfView/Cura-ZOffsetPlugin #35: [Feature request] Reset to original on ;TYPE:SKIN (bottom skin)

j3ffrw, 2023-07-03:
hi,

    Hi I'd like to request a feature in which the ZOffset on apply for the first layer wall(INNER/OUTER) and revert back to original once it is on 1st layer TYPE:SKIN. My use case it to improve the bed adhesion of initial thin lines/wall by squishing it a bit while preventing the SKIN in having ridges due to the offset. 

Regards,
Jeff

```
# 1st layer
G0 F6000 X62.827 Y75.333 Z0.18 ;adjusted by z offset
G92 Z0.2 ;consider this the original z before offset
;TYPE:SKIRT
;TYPE:WALL-INNER
;TYPE:WALL-OUTER
;TYPE:SKIN
G0 Z0.2 ; reset to original for bottom skin
```

fieldOfView, 2023-09-18:
This would create an under- or over-extrusion between the first and second layer.

>  while preventing the SKIN in having ridges due to the offset

Can you elaborate? Why would the skin have ridges when the entire model is offset?

j3ffrw, 2024-02-18:
Hi, in my case i use a negative offiset (-0.02) to ensure the 1st layer walls including the support brim to stick to the bed by squishing it a bit. But if the offset(negative) includes the SKIN the squish causes it to push the filament on the sides of the nozzle creating a ridge. Look like https://www.reddit.com/media?url=https%3A%2F%2Fi.redd.it%2Fmshug77r5mi61.jpg