fieldOfView/Cura-ZOffsetPlugin #38: [Feature Request] Multiple Z-offsets in a grid across the bed to "flatten" the print area with linear extrapolation

KJ7LNW, 2024-01-02:
I was looking at your `_filterGcode` code and it could be easy to add linear extrapolation across a distributed grid of Z-offset samples, because you already adjust all Z values in G0/1 calls from the `adhesion_z_offset` config value.  

The `_filterGcode` function would be extended to support Z-offset extrapolation based on X/Y values: By simply adjusting the Z value of G0/1 calls based on the grid, the printer will do the point-to-point linear extrapolation for us. For example, `adjusted_z = extrapolate(X, Y, Z, ZOffsetSamples)` where `X, Y, Z` are the original G0/1 coordinates and `ZOffsetSamples` is the sample grid.  

More background (mostly copy-pasted [from my stack exchange post](https://3dprinting.stackexchange.com/q/22886/41015)):

We have a [Tronxy X5SA 400 Pro](https://www.tronxy3d.com/collections/x5sa-400-series/products/tronxy-new-version-x5sa-400-pro) which uses a [TR sensor](https://www.tronxy3d.com/products/tronxy-black-tr-auto-leveling-sensor-3d-printer-kit) to "auto-level" the printing bed.  It samples 16 positions across the bed (4 x 4) and provides a +/- level from its initial measurement.  For example, it might produce a grid like this:

```
+0.280   +0.155   -0.108   +0.127   
-0.267   -0.010   -0.005   -0.218   
-0.114   -0.020   +0.004   +0.000   
-0.143   +0.110   +0.253   -0.270  
```

As you can see in this contrived example, measurements toward the center are closer to a Z-offset of 0 because that is what it samples first; toward the edge, it can be off by ~0.3mm.  By turning the sprung leveling screws, I can usually get it to within 0.250mm across the bed.  

However, 0.250mm variance is still too much for 0.1mm-layer prints because at the center it is right on, but on the edge it is 2-3x the layer height above the bed...or worse, in the opposite case, the head will crash if the bed curves in the +Z direction; thus, I have to babysit the first layer and manually +/-Z as needed.

Prints in the center of the bed are great, but adhesion can be poor on the first layer when it prints near the edge, and part thickness can be off by ~1-3 layers.

**Questions:**
- What would be the best way to provide the Z-offset sample grid as a configuration setting to Cura?  A long list of `adhesion_zsample1_1`, `adhesion_zsample1_2`, ... `adhesion_zsample4_4` options seems cumbersome but possible.  Is there is a better way, like reading a 2D array from a JSON file (not that this super UI user-friendly either).
- Other considerations?

fieldOfView, 2024-01-02:
This is something that should really be handled in the firmware (mesh bed leveling; https://marlinfw.org/docs/gcode/G029-mbl.html).

I don't think I want to even try burning my fingers on this. This will lead to multiple different Z heights within one layer, with a high potential of bumping the nozzle into already printer parts while traveling. In short: this plugin is meant to be used "on top of" a (mesh-) leveled bed to adjust adhesion specific for a material.

Feel free to try to convince me otherwise, but for the time being I am closing this as "won't do".

closed 2024-01-02 by fieldOfView

KJ7LNW, 2024-01-02:
> This is something that should really be handled in the firmware (mesh bed leveling; https://marlinfw.org/docs/gcode/G029-mbl.html).

Agreed, and thanks for the link.  Our printer didn't seem to apply the auto-level results, just displays them to the screen. I'm hoping for updated firmware from Tronxy that does. However, maybe I can write a bit of G-code to accomplish the same thing.